gbionics/idyntree #1070: Set resizable window on Windows

dariosortino, 2023-05-03:
As described in https://github.com/ami-iit/lab-events-demos/issues/252#issuecomment-1532538791, at the moment is impossible to resize the visualizer window in Windows OS.

With @S-Dafarra in https://github.com/robotology/human-dynamics-estimation/pull/341/files , we added the possibility to set the height and width of the window in a configuration file as a workaround to this, but it's not convenient if we keep switching between different screens with different resolutions.

cc @traversaro

S-Dafarra, 2023-05-03:
I have tried adding ``setResizable(true)`` but as soon as the visualizer starts, the window shrinks and the visualization is messed up.

Investigating

traversaro, 2023-05-03:
> I have tried adding `setResizable(true)` but as soon as the visualizer starts, the window shrinks and the visualization is messed up.
> 
> Investigating

For testing you could try also to temporary revert https://github.com/robotology/idyntree/pull/976 (by commenting the line https://github.com/robotology/idyntree/blob/v8.1.0/src/visualization/src/Visualizer.cpp#L283) to see if without using SDL the resize on Windows works. If that works, we can think about exposing that so that we can opt-out of using SDL on Windows, as anyhow the problem reported in https://github.com/robotology/idyntree/issues/975 is not a problem if `HumanStateVisualizer` is a process just devoted to visualization .

S-Dafarra, 2023-05-03:
I tried it, but it is still not usable. I think it may be a bug on Windows. I also tried following a bit https://irrlicht.sourceforge.io/forum/viewtopic.php?t=43316&sid=34c5d329ed5118c383d18231a046b4e9, but I had no luck. I am not able to change the size of the window from code.

S-Dafarra, 2023-05-03:
I managed to make it working. Instead of removing [that line](https://github.com/robotology/idyntree/blob/v8.1.0/src/visualization/src/Visualizer.cpp#L283), I had to change it to 
``
irrDevParams.DeviceType = irr::EIDT_WIN32;
``

S-Dafarra, 2023-05-03:
> I tried it, but it is still not usable. I think it may be a bug on the Windows implementation of `irrlicht`. I also tried following a bit https://irrlicht.sourceforge.io/forum/viewtopic.php?t=43316&sid=34c5d329ed5118c383d18231a046b4e9, but I had no luck. I am not able to change the size of the window from code.

Actually, I did a mistake before, and I commented out the wrong lines. Indeed, the suggestion of @traversaro was correct. Now we can decide if we want to resize or close the window :joy:

traversaro, 2023-05-03:
Cool! While we thing how to do this properly @dariosortino I think you can do the local modification suggested in https://github.com/robotology/idyntree/issues/1070#issuecomment-1532723289 and tested by @S-Dafarra in your system/the system used for the demo, so that for future them you can already resize the window, right?

S-Dafarra, 2023-05-03:
> While we thing how to do this properly

What about creating a window on our own and have ``irrlicht`` printing on it? In this way we might be able to deal with the closure more easily. I have been using ``glfw`` in the past and it might also be a quite portable solution. @traversaro what do you think?

S-Dafarra, 2023-05-04:
> Funny story. If I run the test where I use two visualizer, the entire graphical system of my PC crashes

Actually, I was testing an old version in my PC cause I forgot to install :sweat: It seems to be working now

S-Dafarra, 2023-05-05:
With https://github.com/S-Dafarra/idyntree/commit/2ed1e973c99f55e39e8a4a40f80043be052020ce I managed to have mouse inputs working. Hence, I should have recovered all the features. I need to test it on Windows

PR #1071 (2023-05-10): Generate a window manually with GLFW when using irrlicht
This should fix #1070 

I have tested it on Ubuntu 20 and on Windows 11 with a fresh installation and it seems to work. On Windows it works only when using SDL.

I needed to add a good amount of code to process the mouse inputs since with the external window, ``irrlicht`` was not detecting the events anymore.
closed 2023-05-10 by traversaro