gbionics/idyntree #1076: Irrlicht Visualizer not working in MacOS

lrapetti, 2023-05-24:
The visualizer is no longer working in my machine after https://github.com/robotology/idyntree/pull/1071.

e.g. if I try to run `idyntree-model-view` I get the following
```
$ idyntree-model-view -m model.urdf
Irrlicht Engine version 1.8.5
Darwin Kernel Version 20.4.0: Thu Apr 22 21:46:47 PDT 2021; root:xnu-7195.101.2~1/RELEASE_X86_64
2023-05-24 10:48:54.381 idyntree-model-view[93952:12152399] -[GLFWWindow _viewRoot]: unrecognized selector sent to instance 0x7f931cf1b670
zsh: segmentation fault  idyntree-model-view -m 
```

lrapetti, 2023-05-24:
I tried to install `glfw` with `conda install -c conda-forge glfw` as it was added to the documentation in https://github.com/robotology/idyntree/commit/35b0f76a9db3809384e8ebcbdb7cfb11d2cb7a7b but I am getting the same error.

traversaro, 2023-05-24:
Thanks for reporting the issue @lrapetti, @S-Dafarra do you have any idea on this?

lrapetti, 2023-05-24:
Just add something that was not documented, the problem observed in windows in https://github.com/robotology/idyntree/issues/1070 is[was] present also on MacOS

S-Dafarra, 2023-05-24:
Unfortunately, I did not have any MacOS machine to test with. The error is not very informative either.
If there is a setup I can play with, I can give it a try.

I checked online, at that error could appear for a multitude of reasons. At first, we can check if it is related to the detection of the inputs. Maybe we you could try commenting these lines https://github.com/robotology/idyntree/blob/850b379f24e0bf28b95af70d7d14576987128727/src/visualization/src/Visualizer.cpp#L565-L568

At that point mouse inputs will not work, but if that solves the issues, we might narrow down the problem.

Other lines that could be commented could be https://github.com/robotology/idyntree/blob/850b379f24e0bf28b95af70d7d14576987128727/src/visualization/src/Visualizer.cpp#L458

Another thing to check would be to use SDL also on Mac. For that, you could try to change https://github.com/robotology/idyntree/blob/850b379f24e0bf28b95af70d7d14576987128727/src/visualization/src/Visualizer.cpp#L474
to  
```c
#if defined(_WIN32) ||  defined(__APPLE__)
```

S-Dafarra, 2023-05-24:
Also, checking https://gist.github.com/v3n/27e810ac744b076ceeb7, maybe we are missing some lines:
```
#ifdef __APPLE__
  /* We need to explicitly ask for a 3.2 context on OS X */
  glfwWindowHint (GLFW_CONTEXT_VERSION_MAJOR, 3);
  glfwWindowHint (GLFW_CONTEXT_VERSION_MINOR, 2);
  glfwWindowHint (GLFW_OPENGL_FORWARD_COMPAT, GL_TRUE);
  glfwWindowHint (GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);
#endif
```

These lines should be added immediately after https://github.com/robotology/idyntree/blob/850b379f24e0bf28b95af70d7d14576987128727/src/visualization/src/Visualizer.cpp#L458

lrapetti, 2023-05-24:
> Btw, does a Window appear, even just for a split second?

No

Thanks for all the suggestions! I can try them myself and give you feedback

S-Dafarra, 2023-05-24:
> > Btw, does a Window appear, even just for a split second?
> 
> No

This is, in some sense, nice since it may mean that is more related to the creation of a Window with ``glfw`` rather than some internal mechanism of ``irrlicht``. I would suggest to start with https://github.com/robotology/idyntree/issues/1076#issuecomment-1560795151 then

lrapetti, 2023-05-24:
I have just noticed also this warning message in cmake

```
 $cmake .
-- Found YARP: /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.7.2+12-20220919.1+git2b2bff591")
-- Found YARP: /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.7.2+12-20220919.1+git2b2bff591")
-- Found ICUB 1.28.1 (Components: iCubDev ctrlLib skinDynLib iKin iDyn learningMachine perceptiveModels actionPrimitives optimization)
CMake Warning (dev) at /Users/lorenzorapetti/miniforge3/envs/robotologyenv/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (GLFW3) does
  not match the name of the calling package (glfw3).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/StandardFindModule.cmake:104 (find_package_handle_standard_args)
  /Users/lorenzorapetti/Software/robotology-superbuild/build/install/share/YCM/find-modules/Findglfw3.cmake:33 (standard_find_module)
  src/visualization/CMakeLists.txt:80 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
```

S-Dafarra, 2023-05-24:
These lines seem interesting
```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libIrrlicht.dylib             	0x0000000107b08dc6 irr::CIrrDeviceMacOSX::storeMouseLocation() + 150
1   libIrrlicht.dylib             	0x0000000107b089a7 irr::CIrrDeviceMacOSX::run() + 87
2   libIrrlicht.dylib             	0x0000000107ae556b createDeviceEx + 171
3   libidyntree-visualization.dylib	0x00000001055644ef iDynTree::Visualizer::init(iDynTree::VisualizerOptions const&) + 463
4   libidyntree-visualization.dylib	0x000000010556599d iDynTree::Visualizer::addModel(iDynTree::Model const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 93
5   idyntree-model-view           	0x00000001054f06e6 main + 470
6   libdyld.dylib                 	0x00007fff206bdf3d start + 1
```
The offending code seems to be in https://github.com/zaki/irrlicht/blob/master/source/Irrlicht/CIrrDeviceOSX.mm#L1199

I guess it has something to do with 
https://github.com/zaki/irrlicht/blob/master/source/Irrlicht/CIrrDeviceOSX.mm#L1203-L1210

In any case, it seems to be happening in https://github.com/robotology/idyntree/blob/850b379f24e0bf28b95af70d7d14576987128727/src/visualization/src/Visualizer.cpp#L518

traversaro, 2023-05-24:
> If there is a setup I can play with, I can give it a try.

I can set it up.

lrapetti, 2023-05-24:
Using SDL fixed the issue (last suggestion in https://github.com/robotology/idyntree/issues/1076#issuecomment-1560753122).

I will open a PR with the fix! Thanks a lot @S-Dafarra

traversaro, 2023-05-24:
Great!