gbionics/idyntree #1084: Unable to use matlab bindings when installed via conda on Ubuntu 20.04

S-Dafarra, 2023-06-30:
I am trying to use the [robotology one line install](https://github.com/robotology/robotology-superbuild/blob/master/doc/matlab-one-line-install.md) to use the iDynTree wrappers in  Matlab when ``iDynTree`` has not been installed previously. However, I am having the following error:
```
Invalid MEX-file '--/robotology-matlab/mex/iDynTreeMEX.mexa64': /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by --/robotology-matlab/mex/iDynTreeMEX.mexa64)
```
when attempting to use one function of the wrappers.

I have also checked the following 
```
strings /lib/x86_64-linux-gnu/libstdc++.so.6 | grep LIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

and indeed `GLIBCXX_3.4.29` is missing.

I do not have this error when installing the bindings from source.

S-Dafarra, 2023-06-30:
I have tried to set ``export LD_PRELOAD=/lib/x86_64-linux-gnu/libstdc++.so.6`` before launching ``matlab``, but then I have the error 
```
installation finished.
WARNING:
    You currently have a PYTHONPATH environment variable set. This may cause
    unexpected behavior when running the Python interpreter in Mambaforge.
    For best results, please verify that your PYTHONPATH only points to
    directories of packages that are compatible with the Python interpreter
    in Mambaforge: --/robotology-matlab
Installation of mambaforge completed
Installing robotology packages
Traceback (most recent call last):
  File "--/robotology-matlab/condabin/mamba", line 7, in <module>
    from mamba.mamba import main
  File "--/robotology-matlab/lib/python3.10/site-packages/mamba/mamba.py", line 49, in <module>
    import libmambapy as api
  File "--/robotology-matlab/lib/python3.10/site-packages/libmambapy/__init__.py", line 7, in <module>
    raise e
  File "--/robotology-matlab/lib/python3.10/site-packages/libmambapy/__init__.py", line 4, in <module>
    from libmambapy.bindings import *  # noqa: F401,F403
ImportError: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by --/robotology-matlab/lib/python3.10/site-packages/libmambapy/bindings.cpython-310-x86_64-linux-gnu.so)
Installation of robotology packages completed
Creating setup script in --/robotology_setup.m
Deleting mambaforge installer

traversaro, 2023-06-30:
Sorry, I understood the problem was on 22.04, not 20.04 . Then the problem issue is https://github.com/robotology/robotology-superbuild/issues/1159 . The real workaround is to set LD_PRELOAD=$CONDA_PREFIX/lib/libstdc++.so.6, but the problem is that if you are using the one-line install, the `$CONDA_PREFIX` does not exists. Depending on the MATLAB version that you are using, it is possible that the `libstdc++.so.6` shipped with MATLAB itself is new enough, in that case a possible workaround is to undo the usual workaround of renaming `/usr/local/MATLAB/R<RELEASENAME>/sys/os/glnxa64/libstdc++.so.6` to  `/usr/local/MATLAB/R<RELEASENAME>/sys/os/glnxa64/libstdc++.so.6.bak`, so that the `libstdc++.so.6` shipped by MATLAB is used, but this may work only if MATLAB is new enough.

S-Dafarra, 2023-06-30:
> Depending on the MATLAB version that you are using, it is possible that the `libstdc++.so.6` shipped with MATLAB itself is new enough, in that case a possible workaround is to undo the usual [...]

I tried it, but I had the same error. I have R2021b