gbionics/idyntree #1086: How specify `package://` mapping?

markisus, 2023-07-06:
I have a UR10e.urdf that I found in one of the ROS repositories. However, I don't want to use ROS. This urdf uses the `package://` convention, which I'm not familiar with. For example, one mesh location is specified as `package://ur_description/meshes/ur10e/visual/base.dae`. This leads to a RuntimeError when trying to load this urdf: `RuntimeError: /home/mark/manipulation/ur10e.urdf:122: error: URI 'package://ur_description/meshes/ur10e/visual/base.dae' refers to unknown package 'ur_description'`.

The good news is I actually have these meshes on my filesystem. The previously mentioned mesh is located on my computer at `/opt/ros/noetic/share/ur_description/meshes/ur10e/visual/base.dae`. 

How can I communicate this mapping `package://ur_description -> /opt/ros/noetic/share/ur_description` to Drake? Or should I go ahead and manually replace the `package://` URIs with absolute paths inside my urdf?

markisus, 2023-07-06:
Just after posting this, I think I figured it out.
The mapping is specified with the `PackageMap`.

The fix is simply
```
parser = Parser(plant)
parser.package_map().Add("ur_description", "/opt/ros/noetic/share/ur_description/")
```

closed 2023-07-06 by markisus

traversaro, 2023-07-06:
Hello @markisus, are you are sure you were using iDynTree? I think our API is a bit different, that seems Drake's API.

For a reference for anyone having this problem on iDynTree, to permit to iDynTree to resolve `package://` URIs you can pass the `packageDirs` argument to the `load***` methods of `ModelLoader`, see https://robotology.github.io/idyntree/classiDynTree_1_1ModelLoader.html . Alternatively, you can use `GAZEBO_MODEL_PATH`, `ROS_PACKAGE_PATH` or `AMENT_PREFIX_PATH` environment variables. In particular, in you would need to append `/opt/ros/noetic/share` to either `GAZEBO_MODEL_PATH` or `ROS_PACKAGE_PATH`, or `/opt/ros/noetic` to `AMENT_PREFIX_PATH`, see the following code for reference:  https://github.com/robotology/idyntree/blob/v9.1.0/src/model/src/SolidShapes.cpp .

markisus, 2023-07-07:
My apologies. I accidentally ended up on your project by following a link from a Drake thread.

traversaro, 2023-07-07:
> My apologies. I accidentally ended up on your project by following a link from a Drake thread. 

No problem!