gbionics/idyntree #1116: Failed to run the Irrlicht visualizer on Linux

S-Dafarra, 2023-09-26:
Whenever I try to launch a ``iDynTree::Visualizer`` instance on Ubuntu 20.04, I get the following error:

```
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  31 (X_GLXCreateWindow)
  Serial number of failed request:  27
  Current serial number in output stream:  28
Irrlicht Engine version 1.8.4
Linux 5.15.0-84-generic #93~20.04.1-Ubuntu SMP Wed Sep 6 16:15:40 UTC 2023 x86_64
```
Getting into it, I noticed that the error occurs when initializing `irrlicht` in https://github.com/robotology/idyntree/blob/c453ba094a707ab6399eeda740922bdc9cf94a84/src/visualization/src/Visualizer.cpp#L511
and not when creating the window, which is done earlier in
https://github.com/robotology/idyntree/blob/c453ba094a707ab6399eeda740922bdc9cf94a84/src/visualization/src/Visualizer.cpp#L455

Currently, I am using the ``nvidia-driver-525``

S-Dafarra, 2023-09-26:
I noticed that if I comment the lines https://github.com/robotology/idyntree/blob/c453ba094a707ab6399eeda740922bdc9cf94a84/src/visualization/src/Visualizer.cpp#L486-L487
it works.

I am planning to open a PR to use the classical window creation in Linux, while keeping the use of SDL for Windows and Mac

traversaro, 2023-09-26:
Do you think this was a regression of https://github.com/robotology/idyntree/pull/1071 ?

traversaro, 2023-09-26:
> I am planning to open a PR to use the classical window creation in Linux, while keeping the use of SDL for Windows and Mac

SDL or glfw?

S-Dafarra, 2023-09-26:
I think so, although I am not sure why it is happening now. I did test it on Linux back in time. As a mater of fact, I developed it in the very same machine.

> > I am planning to open a PR to use the classical window creation in Linux, while keeping the use of SDL for Windows and Mac
> 
> SDL or glfw?

glfw to create the window and SDL on the irrlicht side. Since on Ubuntu with ``irrlicht`` from apt we cannot use SDL, maybe we can still let ``irrlicht`` create the window on Ubuntu, since there are no resize issues

S-Dafarra, 2023-09-26:
Relevant commit: https://github.com/S-Dafarra/idyntree/commit/6c51ff82e5f5083a77d9f62891110b3c00b75a05

I will test it on Windows too

closed 2023-10-03 by traversaro