gbionics/idyntree #1142: Usage of the estimations done by the library

itsjjaavii, 2024-01-10:
Hello, I’m currently doing an intership at the investigation center of my university and I’ve come across an issue related to the iDynTree library. The thing is that i don’t really understand how to extract information from this library, so I can use it in RL applications. So i wanted to ask you if anyone could guide me on how to begin to do this, if you know a basic tutorial on how to do it. I already found the Github where it explains some aspects of the iDynTree library, however, i still cannot comprehend how to take the estimations calculated by the library and use them. 
Thanks in advance!

traversaro, 2024-01-10:
Which kind of "estimations" you want to use iDynTree for? Which language do you want to use iDynTree in? C++ or Python?

itsjjaavii, 2024-01-10:
Hi, I want to extract from the library the estimated joint torques, CoM and external forces, using Python. The idea is that i could send this estimations through YARP ports to Gazebo, and I know this is possible, however, i do not know which ports i have to use, that's why I'm starting from the iDynTree level.

itsjjaavii, 2024-01-12:
Hello, I also wanted to ask what are the total estimations that the iDynTree library is capable of providing? I know it can provide joint torques and external forces, but I'm not sure if it can provide more. 
@traversaro

traversaro, 2024-01-15:
I am not sure which information are you interested in, but if I understood correctly, note that iDynTree has different classes that serve different purposes, the main ones are:
* [`iDynTree::Model`](https://robotology.github.io/idyntree/classiDynTree_1_1Model.html): Class representing the multibody model information (topology, kinematics parameters, dynamic parameters)
* [`iDynTree::KinDynComputations`](https://robotology.github.io/idyntree/classiDynTree_1_1KinDynComputations.html) class that exposes position forward kinematics (i.e. compute the position of a given frame w.r.t. to the world frame, given the base pose and the internal joint positions), velocity forward kinematics, acceleration forward kinematics, inverse dynamics (i.e. compute the generalized torques given contact external force/torques, robot position, velocities and accelerations. This class is tipically used in controllers or planners.
* [`iDynTree::ExtWrenchesAndJointTorquesEstimator`](https://robotology.github.io/idyntree/classiDynTree_1_1ExtWrenchesAndJointTorquesEstimator.html) class, that estimates external force/torque and joint torques using as an input the the internal joint positions, the robot velocities and the accelerations and the measurement of internal six axis force/torque sensors. This is the class used in the `wholebodydynamics` device that is used on the iCub to estimate joint torques and external contact force/torques.