gbionics/idyntree #1155: Add visualization of a subset of frames in idyntree-model-view

traversaro, 2024-02-21:
It happens quite frequently that you want to visualize how frames are oriented in a model. People are frequently using https://github.com/robotology/whole-body-controllers/blob/90ff965a523f0a120e6a8981b71326c1485e7742/utilities/plotRobotFrames.m, that however is a bit cumbersome to setup. A possible alternative is to add the --frames option to idyntree-model-view tools, something like that:
~~~
idyntree-model-view -m /path/to/folder/model.urdf --frames "head_imu_0 waist_imu_0"
~~~

fyi @Nicogene @S-Dafarra @GiulioRomualdi 

Kind of related to https://github.com/robotology/idyntree/issues/942 .

traversaro, 2024-02-21:
The `idyntree-model-view.cpp` code is quite compact: https://github.com/robotology/idyntree/blob/master/src/tools/idyntree-model-view.cpp .

Nicogene, 2024-02-21:
cc @martinaxgloria

traversaro, 2024-02-21:
Features we can use:
* Visualize frames (added in https://github.com/robotology/idyntree/pull/802)
* Visualize labels for frames (added in https://github.com/robotology/idyntree/pull/879)

S-Dafarra, 2024-03-18:
I have added the possibility to visualize frames of a model in https://github.com/robotology/idyntree/pull/1163/commits/2b82df45bf1b1270c6aa3ea76e8b01652b38f2d2

I can modify ``idyntree-model-view`` too

S-Dafarra, 2024-03-18:
> I can modify `idyntree-model-view` too

Done in https://github.com/robotology/idyntree/pull/1163/commits/fd67ed6a3aabea77673200bed1c25298da7eae82 (Related PR #1163 )

S-Dafarra, 2024-03-18:
Example output

[image]