gbionics/idyntree #1159: Failed to compile in clean environment on Windows

S-Dafarra, 2024-03-07:
I am facing some compilation issue when try to compile from a clean environment.

In particular, if I do
```
mamba install swig=4.1.0 cmake libxml2 eigen numpy
cmake .. -G"Visual Studio 17 2022" -DIDYNTREE_COMPILES_OPTIMALCONTROL=OFF -DIDYNTREE_COMPILES_TOOLS=OFF
cmake --build . --config Release --target INSTALL
```
I have the following error
```
C:\Software\mambaforge\envs\idyntree\Library\include\libxml2\libxml\encoding.h(28,10): error C1083: Cannot open include file: 'iconv.h': No such file or directory [C:\Software\robotology-superbuild\src\iDynTree\build\src\model_io\
xml\idyntree-modelio-xml.vcxproj]
  (compiling source file '../../../../src/model_io/xml/src/XMLParser.cpp')
```
<details>
<summary>Full output</summary>

```
MSBuild version 17.8.5+b5265ef37 for .NET Framework

  1>Checking Build System
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/extern/CMakeLists.txt
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/src/core/CMakeLists.txt
  fpconv.c
  Axis.cpp
  idyntree-private-fpconv.vcxproj -> C:\Software\robotology-superbuild\src\iDynTree\build\extern\idyntree-private-fpconv.dir\Release\idyntree-private-fpconv.lib
  ArticulatedBodyInertia.cpp
  ClassicalAcc.cpp
  Direction.cpp
  InertiaNonLinearParametrization.cpp
  MatrixDynSize.cpp
  GeomVector3.cpp
  Position.cpp
  Rotation.cpp
  RotationalInertia.cpp
  SpatialAcc.cpp
  SpatialForceVector.cpp
  SpatialMomentum.cpp
  SpatialMotionVector.cpp
  SpatialInertia.cpp
  TestUtils.cpp
  Transform.cpp
  TransformDerivative.cpp
  Twist.cpp
  Utils.cpp
  Generating Code...
  Compiling...
  VectorDynSize.cpp
  Wrench.cpp
  PrivateUtils.cpp
  SparseMatrix.cpp
  Triplets.cpp
  CubicSpline.cpp
  SO3Utils.cpp
  Generating Code...
  Auto build dll exports
     Creating library C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-core.lib and object C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-core.exp
  idyntree-core.vcxproj -> C:\Software\robotology-superbuild\src\iDynTree\build\bin\Release\idyntree-core.dll
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/src/model_io/xml/CMakeLists.txt
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/src/model/CMakeLists.txt
  XMLParser.cpp
  ContactWrench.cpp
C:\Software\mambaforge\envs\idyntree\Library\include\libxml2\libxml\encoding.h(28,10): error C1083: Cannot open include file: 'iconv.h': No such file or directory [C:\Software\robotology-superbuild\src\iDynTree\build\src\model_io\
xml\idyntree-modelio-xml.vcxproj]
  (compiling source file '../../../../src/model_io/xml/src/XMLParser.cpp')

  XMLElement.cpp
  DenavitHartenberg.cpp
  XMLAttribute.cpp
  XMLDocument.cpp
  Generating Code...
  FixedJoint.cpp
  ForwardKinematics.cpp
  FreeFloatingState.cpp
  FreeFloatingMatrices.cpp
  Indices.cpp
  Dynamics.cpp
  DynamicsLinearization.cpp
C:\Software\robotology-superbuild\src\iDynTree\src\model\src\DynamicsLinearization.cpp(311,109): warning C4552: '*': result of expression not used [C:\Software\robotology-superbuild\src\iDynTree\build\src\model\idyntree-model.vcxp
roj]
  DynamicsLinearizationHelpers.cpp
  Link.cpp
  LinkState.cpp
  LinkTraversalsCache.cpp
  Jacobians.cpp
  JointState.cpp
  Model.cpp
  ModelTransformers.cpp
  ModelInterfaceDestructors.cpp
  RevoluteJoint.cpp
  PrismaticJoint.cpp
  Generating Code...
  Compiling...
  SolidShapes.cpp
  SubModel.cpp
  Traversal.cpp
  Sensors.cpp
  SixAxisForceTorqueSensor.cpp
  AccelerometerSensor.cpp
  GyroscopeSensor.cpp
  ThreeAxisAngularAccelerometerSensor.cpp
  ThreeAxisForceTorqueContactSensor.cpp
  PredictSensorsMeasurements.cpp
  ModelSensorsTransformers.cpp
  Generating Code...
  Auto build dll exports
     Creating library C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-model.lib and object C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-model.exp
  idyntree-model.vcxproj -> C:\Software\robotology-superbuild\src\iDynTree\build\bin\Release\idyntree-model.dll
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/src/visualization/CMakeLists.txt
  Building Custom Rule C:/Software/robotology-superbuild/src/iDynTree/src/solid-shapes/CMakeLists.txt
  Visualizer.cpp
  InertialParametersSolidShapesHelpers.cpp
  Auto build dll exports
     Creating library C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-visualization.lib and object C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-visualization.exp
  idyntree-visualization.vcxproj -> C:\Software\robotology-superbuild\src\iDynTree\build\bin\Release\idyntree-visualization.dll
  ModelTransformersSolidShapes.cpp
  Generating Code...
  Auto build dll exports
     Creating library C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-solid-shapes.lib and object C:/Software/robotology-superbuild/src/iDynTree/build/lib/Release/idyntree-solid-shapes.exp
  idyntree-solid-shapes.vcxproj -> C:\Software\robotology-superbuild\src\iDynTree\build\bin\Release\idyntree-solid-shapes.dll
```

</details>

<details>
<summary><code>mamba list</code></summary>

```
(idyntree) C:\Software\robotology-superbuild\src\iDynTree\build>mamba list
# packages in environment at C:\Software\mambaforge\envs\idyntree:
#
# Name                    Version                   Build  Channel
bzip2                     1.0.8                hcfcfb64_5    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
intel-openmp              2024.0.0         h57928b3_49841    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
libblas                   3.9.0              21_win64_mkl    conda-forge
libcblas                  3.9.0              21_win64_mkl    conda-forge
libcurl                   8.5.0                hd5e4a3a_0    conda-forge
libexpat                  2.6.1                h63175ca_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libhwloc                  2.9.3           default_haede6df_1009    conda-forge
libiconv                  1.17                 hcfcfb64_2    conda-forge
liblapack                 3.9.0              21_win64_mkl    conda-forge
libsqlite                 3.45.1               hcfcfb64_0    conda-forge
libssh2                   1.11.0               h7dfc565_0    conda-forge
libuv                     1.47.0               hcfcfb64_0    conda-forge
libxml2                   2.12.5               hc3477c8_0    conda-forge
libzlib                   1.2.13               hcfcfb64_5    conda-forge
mkl                       2024.0.0         h66d3029_49657    conda-forge
numpy                     1.26.4          py311h0b4df5a_0    conda-forge
openssl                   3.2.1                hcfcfb64_0    conda-forge
pcre2                     10.40                h17e33f8_0    conda-forge
pip                       24.0               pyhd8ed1ab_0    conda-forge
pthreads-win32            2.9.1                hfa6e2cd_3    conda-forge
python                    3.11.8          h2628c8c_0_cpython    conda-forge
python_abi                3.11                    4_cp311    conda-forge
setuptools                69.1.1             pyhd8ed1ab_0    conda-forge
swig                      4.1.0                h9eeddcc_4    conda-forge
tbb                       2021.11.0            h91493d7_1    conda-forge
tk                        8.6.13               h5226925_1    conda-forge
tzdata                    2024a                h0c530f3_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                hcf57466_18    conda-forge
vc14_runtime              14.38.33130         h82b7239_18    conda-forge
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
```

</details>

cc @traversaro

S-Dafarra, 2024-03-07:
Using the following environment instead, it works:

<details>
<summary><code>mamba list</code></summary>

```
(idyntree_robsub) C:\Software\robotology-superbuild\src\iDynTree\build2>mamba list
# packages in environment at C:\Software\mambaforge\envs\idyntree_robsub:
#
# Name                    Version                   Build  Channel
ace                       7.1.3                h63175ca_2    conda-forge
aiohttp                   3.9.3           py312he70551f_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.1                h63175ca_0    conda-forge
asio                      1.29.0               h63175ca_0    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.16               hec1de76_6    conda-forge
aws-c-cal                 0.6.10               hd481e46_1    conda-forge
aws-c-common              0.9.13               hcfcfb64_0    conda-forge
aws-c-compression         0.2.18               hd481e46_1    conda-forge
aws-c-event-stream        0.4.2                h0f06f08_4    conda-forge
aws-c-http                0.8.1                hdb5aac5_5    conda-forge
aws-c-io                  0.14.5               h08270f9_1    conda-forge
aws-c-mqtt                0.10.2               hfea8755_4    conda-forge
aws-c-s3                  0.5.2                h4b2095a_0    conda-forge
aws-c-sdkutils            0.1.15               hd481e46_1    conda-forge
aws-checksums             0.1.18               hd481e46_1    conda-forge
aws-crt-cpp               0.26.2               h8492d2a_7    conda-forge
aws-sdk-cpp               1.11.267             h93f5800_1    conda-forge
azure-core-cpp            1.11.1               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_1    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_4    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
boost                     1.82.0               hd42ba9a_6    conda-forge
brotli                    1.0.9                hcfcfb64_9    conda-forge
brotli-bin                1.0.9                hcfcfb64_9    conda-forge
bullet-cpp                3.25                 h2ab9e98_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.27.0               hcfcfb64_0    conda-forge
c-compiler                1.7.0                hcfcfb64_0    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                h9b0cee5_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.2.0           py312h0d7def4_0    conda-forge
cppzmq                    4.10.0               h42135b4_1    conda-forge
cxx-compiler              1.7.0                h91493d7_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.1               haebf736_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.4.1                h63175ca_0    conda-forge
double-conversion         3.3.0                h63175ca_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.6.1                h63175ca_0    conda-forge
fcl                       0.7.0                he22821c_4    conda-forge
ffmpeg                    6.1.1           gpl_hb766fab_104    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
[... 265 lines cut ...]
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.5                h63175ca_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.1.1                hcd874cb_0    conda-forge
xorg-libsm                1.2.4                hcd874cb_0    conda-forge
xorg-libx11               1.8.7                hefa74cf_0    conda-forge
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_2    conda-forge
xorg-libxpm               3.5.17               hcd874cb_0    conda-forge
[... cut ...]
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yarl                      1.9.4           py312he70551f_0    conda-forge
zeromq                    4.3.5                h63175ca_1    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

</details>

I wonder which package makes it working

S-Dafarra, 2024-03-07:
Apparently it was enough to add ``pkg-config`` to the list of dependencies to install with ``mamba`` to make it working.

I wonder whether this should be handled otherwise. In any case, I doubt this is an ``iDynTree`` issue.

traversaro, 2024-03-07:
What I do typically is to install all of these when I compile a C/C++ project: `cmake make ninja pkg-config compilers`, exactly as some `find_package` modules of CMake depend on `pkg-config` to be available. I think it could make sense to have a `build-essential` conda-forge metapackage that installs all of it easily, but I never actually went on and open an issue/PR to suggest this.

At the iDynTree-specific level, all the packages necessary for compiling idyntree are documented in https://github.com/robotology/idyntree/blob/master/doc/build-from-source.md#install-dependencies-with-conda-forge, and that includes `pkg-config` (actually the problem there is that swig is not pinned, but that is another story and should be fixed now that https://github.com/conda-forge/glib-feedstock/pull/164 is merged). So I guess on the iDynTree side we are good to close the issue, even if indeed the implicit dependency of cmake on pkg-config remains confusing?

S-Dafarra, 2024-03-07:
> At the iDynTree-specific level, all the packages necessary for compiling idyntree are documented in https://github.com/robotology/idyntree/blob/master/doc/build-from-source.md#install-dependencies-with-conda-forge,

Ah thanks, I did not see this

> So I guess on the iDynTree side we are good to close the issue, even if indeed the implicit dependency of cmake on pkg-config remains confusing?

Yes, sure, thanks!

closed 2024-03-07 by S-Dafarra