gbionics/idyntree #1172: Add option to add names to collision and visual elements in exported URDF

traversaro, 2024-04-03:
@CarlottaSartore (feel free to add more comments) some specific cases when reading URDF in MuJoCo for which it is important to add names to the collision and visual elements exported in URDF. 

The iDynTree URDF exporter should support adding automatically this values, and eventually permit to users to set them to specific values.

traversaro, 2024-04-03:
Actually the code is already there: https://github.com/robotology/idyntree/blob/c6a00762d4bf87a1ecb1fc6655e99369722d2235/src/model_io/codecs/src/URDFModelExport.cpp#L146-L150 . So, the problem is that in creo2urdf we are not setting the collision and visual names, no sure if it could make sense to have an option in the URDF exporter to automatically add collision names if this are not present.

francesco-romano, 2024-04-03:
Yes, it is better to name them univocally if you want to import the model into Mujoco. Note that the name must be unique across visual and collisions (the type is not part of the identifier).
For example we prefix the mesh name with `visual_` and `collision_` when building the model prior to exporting.
I think if two names are equal MuJoCo silently add unnamed geoms when parsing the URDF

francesco-romano, 2024-04-03:
As @traversaro said the possibility to name the meshes is already there and I use it successfully

traversaro, 2024-04-22:
Fixed by https://github.com/robotology/idyntree/pull/1173 .

closed 2024-04-22 by traversaro