gbionics/idyntree #1182: Regression in estimation of redundant external forces since August 2022

traversaro, 2024-05-16:
In https://github.com/robotology/idyntree/pull/1017 there was a regression when the forces are redundant. 

By the way, I suspect the reason of the sharp changes in the estimated FT may be https://github.com/robotology/idyntree/pull/1017 , even if  also `colPivHouseholderQr` should return the least square solution. However, https://github.com/robotology/idyntree/pull/1017 changed a truncated pseudo-inverse for a non-truncated one, something that I missed while reviewing https://github.com/robotology/idyntree/pull/1017 .

> `colPivHouseholderQr` should return the least square solution

Probably it returns the least-square solution in the case of equation then unknown, not more unknown that equations (that is the case now).

S-Dafarra, 2024-05-16:
Hi @traversaro, not sure I understood. In which cases do we need to estimate redundant external wrenches?

traversaro, 2024-05-16:
> Hi @traversaro, not sure I understood. In which cases do we need to estimate redundant external wrenches?

For example when doing `calibStanding` on double support, also known (with unfortunate naming) "macumba calibration".

traversaro, 2024-05-16:
@HosameldinMohamed confirmed that the regression is present.

traversaro, 2024-05-16:
> > Hi @traversaro, not sure I understood. In which cases do we need to estimate redundant external wrenches?
> 
> For example when doing `calibStanding` on double support, also known (with unfortunate naming) "macumba calibration".

Another case is that i you have a robot with an wide patch of skin, and you touch it in many points of the skin, that result in a huge number of small unknown contact forces. You have infinite solutions for explaining the 6D force torque that you estimate for that submodule, and to choose them before the regression we always choose the minimum norm (i.e. least square) solution.