gbionics/idyntree #1187: Unable to get the joint pos limits in python

GiulioRomualdi, 2024-05-23:
I wrote the following code to get the joint position limits in python

```python
def get_joint_limits(kindyn: idyn.KinDynComputations, joint_list: List[str]):
    if not kindyn.isValid():
        raise RuntimeError("The KinDynComputations object is not valid")

    lower_limits = np.array([-np.pi] * len(joint_list))
    upper_limits = np.array([np.pi] * len(joint_list))

    for i in range(kindyn.model().getNrOfJoints()):
        joint = kindyn.model().getJoint(i)

        if not joint.hasPosLimits():
            continue

        for dof in range(joint.getNrOfDOFs()):
            upper_lim = 0.0
            lower_lim = 0.0

            if not joint.getPosLimits(dof, lower_lim, upper_lim):
                continue

            upper_limits[joint.getDOFsOffset() + dof] = upper_lim
            lower_limits[joint.getDOFsOffset() + dof] = lower_lim

    return lower_limits, upper_limits
```

however when I called I got the following error
```
File "/home/gromualdi/robot-code/element_rl-for-locomotion/code/ergocub_amp_walking/config/robots/ergoCubGazeboV1_1/../../../script/amp_walking.py", line 125, in get_joint_limits
    if not joint.getPosLimits(dof, lower_lim, upper_lim):
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/idyntree/swig.py", line 3625, in getPosLimits
    return _iDynTree.IJoint_getPosLimits(self, _index, min, max)
TypeError: in method 'IJoint_getPosLimits', argument 3 of type 'double &'
```

cc @traversaro @Giulero

traversaro, 2024-05-23:
`getPosLimits` and other methods that uses primitive types passed via reference as outputs can't be easily wrapped in Python or MATLAB, probably we should suppress them to avoid that they appear in SWIG bindings at all. As an alternative, you can use `getMinPosLimit()` and `getMaxPosLimit` methods.

traversaro, 2024-05-23:
See for example:
* https://github.com/robotology/blender-robotics-utils/blob/8839079844acb6c3a91860276b083a9a06a4be4c/script/urdfToBlender/urdfToBlender.py#L211
* https://github.com/search?q=org%3Aami-iit+getMaxPosLimit++language%3APython&type=code&l=Python

GiulioRomualdi, 2024-05-24:
Thank you, @traversaro! The methods you suggested work fine!

GiulioRomualdi, 2024-06-25:
I think we can close this issue

closed 2024-06-25 by GiulioRomualdi