gbionics/idyntree #1189: iDynTree PyPI packages need to be rebuilt against NumPy 2.0

diegoferigo, 2024-06-17:
In a downstream project, overnight the CI pipeline started to fail with the following error:

```
ImportError: 
A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.0 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.
```

I guess that the wheels distributed through PyPI need to be rebuilt against NumPy 2.0. The good part is that new wheels should be backward compatible with NumPy 1.0, therefore there's no need to enforce NumPy 2.0 on the project level. It can be forced to be >2.0 only in CI/CD.

traversaro, 2024-06-17:
Probably we can just do a new release that will be built against numpy 2 ?

traversaro, 2024-06-17:
For reference, the conda packages were migrated for numpy 2 compatibility in https://github.com/conda-forge/idyntree-feedstock/pull/91 .

diegoferigo, 2024-06-17:
> Probably we can just do a new release that will be built against numpy 2 ?

Yep, and I guess that we also need to remove support of building wheels on Python 3.7 and 3.8 since it seems that there are no wheels for those versions:

https://pypi.org/project/numpy/2.0.0/#files

closed 2024-06-17 by traversaro