gbionics/idyntree #1201: ModelExporter: exporting a frame attached to the base link, make that fake link the base link

Nicogene, 2024-08-29:
@mfussi66 @kaidegast and I noticed this strange behaviour when trying to export frames attached to the link that is passed as `baseLink` to the `ModelExporterOptions`.
Basically what happens is that the frame exported become the new baseLink and the fixed joint between the base link and the frame is reversed (the frame is the parent, the base link is the child).

Here is the python snipped I used for testing this strange behaviour:

```python
import idyntree.swig as idyn

model = idyn.Model()
# Let's add the base link
model.addLink("root_link",idyntree.swig.Link())
# Let's add a frame to the base link
model.addAdditionalFrameToLink("root_link", "root_link_frame", idyn.Transform.Identity())
export_options = idyn.ModelExporterOptions()
export_options.robotExportedName = "foo"
export_options.baseLink = "root_link"
model_exporter= idyntree.swig.ModelExporter()
model_exporter.init(model)
model_exporter.setExportingOptions(export_options)
model_exporter.exportModelToFile("model.urdf")
```

And here is the resulting URDF:
```urdf
<?xml version="1.0"?>
<robot name="foo">
  <link name="root_link_frame"/>
  <joint name="root_link_frame_fixed_joint" type="fixed">
    <origin xyz="0 0 0" rpy="0 -0 0"/>
    <parent link="root_link_frame"/>
    <child link="root_link"/>
  </joint>
  <link name="root_link">
    <inertial>
      <mass value="0"/>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <inertia ixx="5.981524883892869e-154" ixy="6.013469529641847e-154" ixz="1.7174754640937156e-259" iyy="9.635107555939643e+140" iyz="1.9608652923211385e+243" izz="-4.538015473251081e+279"/>
    </inertial>
  </link>
</robot>
```

cc @traversaro

traversaro, 2024-09-02:
The `exportFirstBaseLinkAdditionalFrameAsFakeURDFBase` option of `ModelExporterOptions` (https://robotology.github.io/idyntree/structiDynTree_1_1ModelExporterOptions.html#a776c8f98ae5c58e038e3a1962f9fd318) controls this behavior. The history behind it is a bit convoluted, but the basic reason is that one of the URDF parser used by ROS has a bug that basically ignores the mass/inertia of the root link (see https://github.com/ros/kdl_parser/issues/27), so having  `exportFirstBaseLinkAdditionalFrameAsFakeURDFBase`  set to `true` by default make sures that if a the model has a fake link in the base, when loaded in kdl the root link is not ignored. 

However, if you need to avoid this behaviour, you can set it to `False`:

~~~python
export_options.exportFirstBaseLinkAdditionalFrameAsFakeURDFBase = False
~~~

traversaro, 2024-09-02:
fyi @kaidegast @mfussi66 @Nicogene

Nicogene, 2024-09-02:
> The `exportFirstBaseLinkAdditionalFrameAsFakeURDFBase` option of `ModelExporterOptions` ([robotology.github.io/idyntree/structiDynTree_1_1ModelExporterOptions.html#a776c8f98ae5c58e038e3a1962f9fd318](https://robotology.github.io/idyntree/structiDynTree_1_1ModelExporterOptions.html#a776c8f98ae5c58e038e3a1962f9fd318)) controls this behavior. The history behind it is a bit convoluted, but the basic reason is that one of the URDF parser used by ROS has a bug that basically ignores the mass/inertia of the root link (see [ros/kdl_parser#27](https://github.com/ros/kdl_parser/issues/27) and [ros/robot_model#6](https://github.com/ros/robot_model/issues/6)), so having `exportFirstBaseLinkAdditionalFrameAsFakeURDFBase` set to `true` by default make sures that if a the model has a fake link in the base, when loaded in kdl the root link is not ignored.
> 
> However, if you need to avoid this behaviour, you can set it to `False`:
> 
> ```python
> export_options.exportFirstBaseLinkAdditionalFrameAsFakeURDFBase = False
> ```

Thanks @traversaro! Do you think it is safe to set it to `false` in `creo2urdf`?

traversaro, 2025-06-20:
fyi @FabioBergonti

closed 2025-06-20 by Nicogene