gbionics/idyntree #1210: Idyntree-model-view fails on WSL

lorycontixd, 2024-10-09:
Hi, I have WSL (Ubuntu 22.04 LTS) installed on Windows 11 with the following specs:

```
WSL version: 2.3.24.0
Kernel version: 5.15.153.1-2
WSLg version: 1.0.65
MSRDC version: 1.2.5620
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
```

By running the command ```idyntree-model-view -m <model_urdf>```, i get the following error:

```
Irrlicht Engine version 1.8.5
SDL Version 1.2.68
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  148 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  117
  Current serial number in output stream:  117
```

traversaro, 2024-10-09:
Can you try to create a clean environment with `conda create -n testidyntree idyntree icub-models` and run:

~~~
conda activate testidyntree
idyntree-model-view -m $CONDA_PREFIX/share/iCub/robots/iCubGazeboV2_7/model.urdf
~~~

to understand if the problem is env-specific or due to your system? In my system with Ubuntu 24.04 (not Ubuntu 22.04) this works fine:

~~~
(testidyntree) traversaro@IITBMP014LW012:~$ conda list
# packages in environment at /home/traversaro/miniforge3/envs/testidyntree:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
assimp                    5.4.3                h8943939_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
expat                     2.6.3                h5888daf_0    conda-forge
gettext                   0.22.5               he02047a_3    conda-forge
gettext-tools             0.22.5               he02047a_3    conda-forge
glfw                      3.4                  hd590300_0    conda-forge
icu                       75.1                 he02047a_0    conda-forge
icub-models               2.8.0           py313h46c70d0_0    conda-forge
idyntree                  13.0.0          py313hf2840b1_1    conda-forge
ipopt                     3.14.16             h122424a_10    conda-forge
irrlicht                  1.8.5                hcce6d95_5    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
ld_impl_linux-64          2.43                 h712a8e2_1    conda-forge
libasprintf               0.22.5               he8f35ee_3    conda-forge
libasprintf-devel         0.22.5               he8f35ee_3    conda-forge
libblas                   3.9.0           24_linux64_openblas    conda-forge
libboost                  1.86.0               hb8260a3_2    conda-forge
libcap                    2.69                 h0f662aa_0    conda-forge
libcblas                  3.9.0           24_linux64_openblas    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libexpat                  2.6.3                h5888daf_0    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libflac                   1.4.3                h59595ed_0    conda-forge
libgcc                    14.1.0               h77fa898_1    conda-forge
libgcc-ng                 14.1.0               h69a702a_1    conda-forge
libgcrypt                 1.11.0               h4ab18f5_1    conda-forge
libgettextpo              0.22.5               he02047a_3    conda-forge
libgettextpo-devel        0.22.5               he02047a_3    conda-forge
libgfortran               14.1.0               h69a702a_1    conda-forge
libgfortran-ng            14.1.0               h69a702a_1    conda-forge
libgfortran5              14.1.0               hc5f4f2c_1    conda-forge
libgl                     1.7.0                ha4b6fd6_1    conda-forge
libglib                   2.82.1               h2ff4ddf_0    conda-forge
libglu                    9.0.0             ha6d2627_1004    conda-forge
libglvnd                  1.7.0                ha4b6fd6_1    conda-forge
libglx                    1.7.0                ha4b6fd6_1    conda-forge
libgomp                   14.1.0               h77fa898_1    conda-forge
libgpg-error              1.50                 h4f305b6_0    conda-forge
libhwloc                  2.11.1          default_hecaa2ac_1000    conda-forge
libiconv                  1.17                 hd590300_2    conda-forge
libjpeg-turbo             3.0.0                hd590300_1    conda-forge
liblapack                 3.9.0           24_linux64_openblas    conda-forge
libmpdec                  4.0.0                h4bc722e_0    conda-forge
libogg                    1.3.5                h4ab18f5_0    conda-forge
libopenblas               0.3.27          pthreads_hac2b453_1    conda-forge
libopus                   1.3.1                h7f98852_1    conda-forge
libosqp                   0.6.3                h59595ed_0    conda-forge
libpng                    1.6.44               hadc24fc_0    conda-forge
libqdldl                  0.1.5                h27087fc_1    conda-forge
[... 24 lines cut ...]
pthread-stubs             0.4               hb9d3cd8_1002    conda-forge
pulseaudio-client         17.0                 hb77b528_0    conda-forge
python                    3.13.0          h9ebbce0_100_cp313    conda-forge
python_abi                3.13                    5_cp313    conda-forge
readline                  8.2                  h8228510_1    conda-forge
sdl                       1.2.68               h293081c_0    conda-forge
sdl2                      2.30.7               h3ed165c_0    conda-forge
tk                        8.6.13          noxft_h4845f30_101    conda-forge
tzdata                    2024b                hc8b5060_0    conda-forge
unixodbc                  2.3.12               h661eb56_0    conda-forge
[... cut ...]
xorg-libxinerama          1.1.5                h5888daf_1    conda-forge
xorg-libxxf86vm           1.1.5                hb9d3cd8_3    conda-forge
xorg-xextproto            7.3.0             hb9d3cd8_1004    conda-forge
xorg-xorgproto            2024.1               hb9d3cd8_1    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
~~~

traversaro, 2024-10-09:
Can you also report the output of `env` in wsl?

traversaro, 2024-10-09:
I just tried with Ubuntu 22.04 in WSL, and I was able to reproduce your problem.

lorycontixd, 2024-10-09:
Below is the output for ```env```:

<details><summary>env output</summary>
<p>

```
HOSTTYPE=x86_64
LANG=en_US.UTF-8
PATH=/home/lorenzoconti/mambaforge3/bin:/home/lorenzoconti/miniforge3/condabin:/opt/ros/humble/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v12.4/bin:/mnt/c/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v12.4/libnvvp:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/MATLAB/R2024a/runtime/win64:/mnt/c/Program Files/MATLAB/R2024a/bin:/mnt/c/Program Files/Microsoft SQL Server/150/Tools/Binn/:/mnt/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn/:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/NVIDIA Corporation/Nsight Compute 2024.1.0/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Program Files/NVIDIA Corporation/NVIDIA NvDLISR:/mnt/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit/:/mnt/c/Program Files/Docker/Docker/resources/bin:/mnt/c/Users/lconti/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/lconti/AppData/Local/Programs/Microsoft VS Code/bin:/mnt/c/Users/lconti/.dotnet/tools:/mnt/c/Users/lconti/AppData/Local/gitkraken/bin:/mnt/c/Users/lconti/.dotnet/tools:/home/lorenzoconti/.pixi/bin
TERM=xterm-256color
XDG_RUNTIME_DIR=/run/user/1000/
DISPLAY=:0
WAYLAND_DISPLAY=wayland-0
PULSE_SERVER=unix:/mnt/wslg/PulseServer
WSL2_GUI_APPS_ENABLED=1
WSLENV=
WSL_INTEROP=/run/WSL/246_interop
NAME=IITICUBLAP257
HOME=/home/lorenzoconti
USER=lorenzoconti
LOGNAME=lorenzoconti
SHELL=/usr/bin/zsh
WSL_DISTRO_NAME=Ubuntu-22.04
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
SHLVL=1
PWD=/mnt/c/Users/lconti
OLDPWD=/mnt/c/Users/lconti
ZSH=/home/lorenzoconti/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
ROS_DISTRO=humble
ROS_LOCALHOST_ONLY=0
ROS_PYTHON_VERSION=3
ROS_VERSION=2
LIBGL_ALWAYS_INDIRECT=0
CONDA_EXE=/home/lorenzoconti/miniforge3/bin/conda
_CE_M=
_CE_CONDA=
CONDA_PYTHON_EXE=/home/lorenzoconti/miniforge3/bin/python
CONDA_SHLVL=0
_=/usr/bin/env
```

</p>
</details>

traversaro, 2024-10-09:
I did a bit of tests:
* conda binary on WSL with Ubuntu 24.04 : Works ✅ 
* conda binary on WSL with Ubuntu 22.04 : Fails ❌
* build from source on WSL with Ubuntu 22.04 apt dependencies : Works ✅ 
* build from source on WSL with Ubuntu 22.04 with conda-forge dependencies : Fails ❌

So there is some strange interaction between sdl/irrlicht from conda-forge and WSL/Ubuntu 22.04. Thanks a lot for reporting, but it seems intricate, so I am not sure I will be able to look into this. Just to understand, could you use Ubuntu 24.04 in WSL instead of 22.04 as a workaround? You just need to install it from the Microsoft Store, and it should be quite easy to setup.

lorycontixd, 2024-10-09:
I made it much simpler and moved onto an Ubuntu 24.04 partition 🥇 
Just wanted to alert you of this problem happening on Ubuntu 22.04.
Thanks for the support.

flferretti, 2024-12-20:
I've also encountered this problem using Wayland. The solution for me was to compile idyntree with wayland support as in https://github.com/robotology/idyntree/compare/master...flferretti:idyntree:wayland_support. @traversaro do you think we could add this? One main problem that I see is that then on conda-forge there would be two different recipes, one for X11 and one for Wayland as I couldn't find a way to check the session apart from using `$XDG_SESSION_TYPE`.

FYI @xela-95

traversaro, 2024-12-20:
I still I am not sure how to replicate the problem. You are running in a wayland system in which XWayland is not running?

flferretti, 2024-12-22:
I am running on Ubuntu 24.10 (Wayland). XWayland should be running as `xeyes` works

traversaro, 2024-12-22:
Ack! It is still not clear to me what is happening, but perhaps we can just try first to open the window with X11, and if it fails try with Wayland? Both will link both X11 and Wayland, but that is not a problem I guess.

S-Dafarra, 2024-12-23:
I wonder if it is related to https://github.com/robotology/idyntree/issues/1116, but in theory it should be using SDL: https://github.com/conda-forge/irrlicht-feedstock/blob/d7f0e13c54a887efe13588402859f83b80591c29/recipe/build.sh#L9

flferretti, 2024-12-23:
> Ack! It is still not clear to me what is happening, but perhaps we can just try first to open the window with X11, and if it fails try with Wayland? Both will link both X11 and Wayland, but that is not a problem I guess.

In https://github.com/robotology/idyntree/commit/68fb6760c75f590361586604339b70b222fee2ca I added X11 as a fallback for Wayland. We can also do the opposite, but since from Ubuntu 24.10 wayland is selected by default, this option might be a bit more future-oriented

traversaro, 2024-12-23:
That seems good to me, the only doubt is that if the glfw versions that we use is compiled with wayland support, however if you open a PR we should easily detect that as we would have a linking error otherwise.

traversaro, 2024-12-23:
At a first glance, the wayland support in apt package of glfw3 is not available in 20.04, 22.04 and 24.04 https://packages.ubuntu.com/focal/libglfw3 and https://packages.ubuntu.com/jammy/libglfw3, while it is available from 24.10 ( https://packages.ubuntu.com/oracular/libglfw3). So we can't just enable that by default as it will result in a linking error. Let's check if there is some macro in glfw to check if wayland support is enabled or not.

traversaro, 2024-12-23:
Unfortunately there is no macro to detect if glfw3 was compiled with wayland support or not. Perhaps the easist solution is just to have an option to enable or not wayland support (with X11 fallback), and set its default to `ON` if CONDA_PREFIX is defined and `OFF` otherwise.

PR #1223 (2025-01-10): Enable runtime window system selection on Linux in `iDynTree::Visualizer`
This PR introduces support for Wayland in the `iDynTree::Visualizer`, allowing it to use Wayland surfaces and fallback to X11 when necessary.

Fixes #1210
closed 2025-01-10 by traversaro