gbionics/idyntree #1217: Tests on Python bindings fail with `SEGFAULT` when `PYTHONWARNINGS=error` is set

flferretti, 2024-12-04:
When running the tests with `PYTHONWARNINGS='error' pixi run test`, I get segmentation faults on `PythonGeomtryTests`, `PythonJointTests`, `PythonHelperTests`, `PythonModelLoaderTests` and `PythonExtWrenchesAndJointTorquesEstimatorUnitTest`. By re-running the failed tests in verbose mode I get:

<details><summary><code>PYTHONWARNINGS='error' pixi run test --rerun-failed --output-on-failure</code></summary>
<p>

```make
conda:jaxsim
✨ Pixi task (configure): cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$CMAKE_INSTALL_PREFIX -DBUILD_TESTING:BOOL=ON -DIDYNTREE_USES_PYTHON:BOOL=ON -DIDYNTREE_USES_PYTHON_PYBIND11:BOOL=ON -DIDYNTREE_DETECT_ACTIVE_PYTHON_SITEPACKAGES:BOOL=ON -G Ninja -S . -B .build
-- Found YARP: /home/fferretti-iit.local/git/idyntree/.pixi/envs/default/lib/cmake/YARP (found version "3.9.0~dev")
-- Found YCM: /home/fferretti-iit.local/git/idyntree/.pixi/envs/default/share/cmake/YCM (found version "0.16.9")
-- Found YARP: /home/fferretti-iit.local/git/idyntree/.pixi/envs/default/lib/cmake/YARP (found version "3.9.0~dev")
-- Linking SWIG Python bindings against NumPy '2.1.2'
-- Found pybind11: /home/fferretti-iit.local/git/idyntree/.pixi/envs/default/include (found version "2.13.6")
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done (0.5s)
CMake Warning (dev) at .pixi/envs/default/share/cmake-3.30/Modules/UseSWIG.cmake:945 (add_library):
  The library that is being linked to, idyntree-sensors, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  Linking the iDynTree::idyntree-sensors target is deprecated, please link
  iDynTree::idyntree-model that provides the symbols formerly containd in
  iDynTree::idyntree-sensors

Call Stack (most recent call first):
  bindings/python/CMakeLists.txt:10 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at .pixi/envs/default/share/cmake-3.30/Modules/FindPython/Support.cmake:4239 (add_library):
  The library that is being linked to, idyntree-sensors, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  Linking the iDynTree::idyntree-sensors target is deprecated, please link
  iDynTree::idyntree-model that provides the symbols formerly containd in
  iDynTree::idyntree-sensors

Call Stack (most recent call first):
  .pixi/envs/default/share/cmake-3.30/Modules/FindPython3.cmake:606 (__Python3_add_library)
  .pixi/envs/default/share/cmake/pybind11/pybind11NewTools.cmake:269 (python3_add_library)
  bindings/pybind11/CMakeLists.txt:1 (pybind11_add_module)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at .pixi/envs/default/share/cmake-3.30/Modules/UseSWIG.cmake:945 (add_library):
  The library that is being linked to, idyntree-sensors, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  Linking the iDynTree::idyntree-sensors target is deprecated, please link
  iDynTree::idyntree-model that provides the symbols formerly containd in
  iDynTree::idyntree-sensors

Call Stack (most recent call first):
  bindings/python/CMakeLists.txt:10 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at .pixi/envs/default/share/cmake-3.30/Modules/FindPython/Support.cmake:4239 (add_library):
  The library that is being linked to, idyntree-sensors, is marked as being
  deprecated by the owner.  The message provided by the developer is:

  Linking the iDynTree::idyntree-sensors target is deprecated, please link
  iDynTree::idyntree-model that provides the symbols formerly containd in
  iDynTree::idyntree-sensors

Call Stack (most recent call first):
  .pixi/envs/default/share/cmake-3.30/Modules/FindPython3.cmake:606 (__Python3_add_library)
  .pixi/envs/default/share/cmake/pybind11/pybind11NewTools.cmake:269 (python3_add_library)
  bindings/pybind11/CMakeLists.txt:1 (pybind11_add_module)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at .pixi/envs/default/share/cmake-3.30/Modules/UseSWIG.cmake:945 (add_library):
  The library that is being linked to, idyntree-sensors, is marked as being
  deprecated by the owner.  The message provided by the developer is:
[... 333 lines cut ...]
# Option to filter deprecation warnings
option(IDYNTREE_FILTER_DEPRECATION_WARNINGS "Filter out deprecation warnings" FALSE)

# Remove deprecation warnings for a specific part of iDynTree
macro(idyntree_disable_deprecation_warnings)
  if(IDYNTREE_FILTER_DEPRECATION_WARNINGS)
    if(MSVC)
      add_definitions(/wd4996)
    else()
      CHECK_CXX_COMPILER_FLAG("-Wno-deprecated" CXX_HAS_WNO_DEPRECATED)
      if(CXX_HAS_WNO_DEPRECATED)
        add_definitions(-Wno-deprecated)
      endif()
    endif()
  endif()
endmacro()
```

Other issues mentioning this problem in SWIG 0.4.3:

- https://github.com/swig/swig/issues/3061
- https://github.com/swig/swig/issues/2881
- https://github.com/pymupdf/PyMuPDF/issues/3931
- https://github.com/AMICI-dev/AMICI/issues/2565
- https://github.com/lballabio/QuantLib-SWIG/issues/647

traversaro, 2024-12-04:
I did not tested, but I am afraid that `-Wno-deprecated` will suppress C++ deprecation warnings, not Python ones.

jesper-friis, 2025-02-19:
I also see this issue. I think the reason for this segfault is that SWIG assumes that `PyType_FromSpec()` never returns NULL when called from `swig_varlink_type()` with SWIG_HEAPTYPES defined.

However, with Python 3.11 (and probably earlier version as well) `PyType_FromSpec()` issues a `DeprecationWarning: builtin type swigvarlink has no __module__ attribute`. With `PYTHONWARNINGS=error` this warning is turned into an error which causes `PyType_FromSpec()` to return NULL.

This NULL is then returned by `swig_varlink_type()` and causes a segfault in `SWIG_Python_newvarlink()`:

```C
/* Create a variable linking object for use later */
SWIGINTERN PyObject *
SWIG_Python_newvarlink(void) {
  swig_varlinkobject *result = PyObject_New(swig_varlinkobject, swig_varlink_type());
  if (result) {
    result->vars = 0;
  }
  return ((PyObject*) result);
}
```

when it calls `PyObject_New()` with a NULL pointer as second argument.

The correct fix seems for this issue seems that `swig_varlink_type()` should define a `__module__` slot.

However, this will not prevent SWIG causing segfaults in the case of future DeprecationWarnings from `PyType_FromSpec()`.