gbionics/idyntree #1222: Python bindings fail when importing idyntree in python

HosameldinMohamed, 2024-12-19:
## Version

`v13.2.0`

## How to reproduce the issue

```python
≻ ipython
Python 3.12.2 | packaged by conda-forge | (main, Feb 16 2024, 20:50:58) [GCC 12.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.25.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import yarp

In [2]: import idyntree
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[2], line 1
----> 1 import idyntree

File ~/code/robotology-superbuild/build/install/lib/python3.12/site-packages/idyntree/__init__.py:1
----> 1 from . import swig
      2 from . import visualize
      3 from . import bindings

File ~/code/robotology-superbuild/build/install/lib/python3.12/site-packages/idyntree/swig.py:10
      8 # Import the low-level C/C++ module
      9 if __package__ or "." in __name__:
---> 10     from . import _iDynTree
     11 else:
     12     import _iDynTree

ImportError: /home/Hosameldin/code/robotology-superbuild/build/install/lib/python3.12/site-packages/idyntree/_iDynTree.so: undefined symbol: _ZN8iDynTree31removeAdditionalFramesFromModelERKNS_5ModelERS0_St6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EE

In [3]: 

```

It seems the issue is related to https://github.com/robotology/idyntree/pull/1219

traversaro, 2024-12-19:
Is this a clean build? Are you sure that you do not disabled Python bindings and you have an old and outdated version of the python bindings of idyntree installed?

HosameldinMohamed, 2024-12-20:
> Is this a clean build? Are you sure that you do not disabled Python bindings and you have an old and outdated version of the python bindings of idyntree installed?

Yes it was a clean build of `robotology-superbuild` using dependencies from conda, the option `ROBOTOLOGY_USES_PYTHON` is set to ON, and as a result `IDYNTREE_USES_PYTHON` is set to ON

traversaro, 2024-12-21:
> using dependencies from conda,

Can you report your environment by copy&pasting the output of `conda list` and `conda info` ?

traversaro, 2024-12-21:
Are you sure that you do not have `idyntree` installed also in the `conda` environment?

HosameldinMohamed, 2024-12-23:
> Are you sure that you do not have `idyntree` installed also in the `conda` environment?

You're right! I did find another `idyntree` package installed by `conda` in my system. I guess some time ago I tried installing a package that installed also `idyntree` as a dependency.

Once I removed it, this issue disappeared!

Thanks a lot, and sorry for the false alarm!

closed 2024-12-23 by HosameldinMohamed