gbionics/idyntree #1246: LinkContactWrenches.addNewContactInFrame() not working fine in python

LoreMoretti, 2025-06-09:
The `LinkContactWrenches.addNewContactInFrame()` method added by https://github.com/robotology/idyntree/pull/1245 seems to not work fine in python.

I have tried to replicate the c++ test of https://github.com/robotology/idyntree/blob/master/src/model/tests/ContactWrenchUnitTest.cpp in Python with the following snippet of code:

```
import idyntree.bindings as iDynTree
import numpy as np

class OneLinkModel:
    def __init__(self):
        self.m_model = iDynTree.Model()
        self.m_linkName = "link"

        # Create a model with one link
        link = iDynTree.Link()
        self.m_model.addLink(self.m_linkName, link)

    def addContactWrench(self):
        frameName = "frame"
        x = 2.0  # X coordinate of the frame

        # Add a frame to the link for the contact wrench
        link_H_frame = iDynTree.Transform()
        link_H_frame.setRotation(iDynTree.Rotation.Identity())
        link_H_frame.setPosition(iDynTree.Position(x, 0.0, 0.0))
        self.m_model.addAdditionalFrameToLink(self.m_linkName, frameName, link_H_frame)

        # Create contact wrench structures
        contactWrenches = iDynTree.LinkContactWrenches(self.m_model)

        contactWrench = iDynTree.ContactWrench()
        contactWrench.contactPoint = iDynTree.Position(x, 0.0, 0.0)

        fz = 10.0
        force = np.array([0.0, 0.0, fz])
        torque = np.array([0.0, 0.0, 0.0])
        contactWrench.contactWrench = iDynTree.Wrench(force, torque)

        contactWrenches.addNewContactInFrame(
            self.m_model,
            self.m_model.getFrameIndex(frameName),
            contactWrench
        )

        netWrenches = iDynTree.LinkWrenches(self.m_model)
        contactWrenches.computeNetWrenches(netWrenches)

        netWrench = netWrenches[self.m_model.getLinkIndex(self.m_linkName)]

        print("Net Wrench:", netWrench.toNumPy())

        # Check linear force components
        assert abs(netWrench.toNumPy()[0] - 0.0) < 1e-6
        assert abs(netWrench.toNumPy()[1] - 0.0) < 1e-6
        assert abs(netWrench.toNumPy()[2] - fz) < 1e-6

        # Check torque components
        assert abs(netWrench.toNumPy()[0] - 0.0) < 1e-6
        assert abs(netWrench.toNumPy()[1] + x * fz) < 1e-6
        assert abs(netWrench.toNumPy()[2] - 0.0) < 1e-6

if __name__ == "__main__":
    oneLinkModel = OneLinkModel()
    oneLinkModel.addContactWrench()
```

The assert statement  `assert abs(netWrench.toNumPy()[1] + x * fz) < 1e-6` fails, because `netWrench.toNumPy()` stores just zeros.

traversaro, 2025-06-09:
Can you try (as a test) to add a `iDynTree::Wrench getWrench(const LinkIndex link)` method to `LinkWrenches`, regenerate the bindings (if necessary doing a small modificatin to `idyntree.i`) and check if instead:

~~~
netWrench = netWrenches.getWrench(self.m_model.getLinkIndex(self.m_linkName)]) 
~~~ 

works?

LoreMoretti, 2025-06-09:
> Can you try (as a test) to add a iDynTree::Wrench getWrench(const LinkIndex link) method to LinkWrenches, regenerate the bindings (if necessary doing a small modificatin to idyntree.i) and check if instead:
> 
> ```
>   netWrench = netWrenches.getWrench(self.m_model.getLinkIndex(self.m_linkName)]) 
> ```

Yes, but given the result of https://github.com/robotology/idyntree/issues/1246#issuecomment-2956296134 maybe the issue is before this line?

traversaro, 2025-06-09:
> > Can you try (as a test) to add a iDynTree::Wrench getWrench(const LinkIndex link) method to LinkWrenches, regenerate the bindings (if necessary doing a small modificatin to idyntree.i) and check if instead:
> > 
> > ```
> >   netWrench = netWrenches.getWrench(self.m_model.getLinkIndex(self.m_linkName)]) 
> > ```
> 
> Yes, I can try! But given the result of https://github.com/robotology/idyntree/issues/1246#issuecomment-2956296134 maybe the issue is before this line?

Yes, you are right.

traversaro, 2025-06-09:
Can you add some prints also in the C++ part? One thing that I noticed is that probably we are missing a rotation on the wrench part.

closed 2025-06-10 by GiulioRomualdi